polarfire-soc/polarfire-soc-bare-metal-library #2: Periodic IRQ Example with Icicle Kit (Rev 1.0)

Jesse-Millwood, 2020-12-09:
### Background
I am trying to get the mpfs periodic irq example to run on my Icicle Kit over JTAG. 
My kit came with a Flash Pro 5 so I have J9 open and have the Flash Pro 5 connected to the JTAG connector. 

### Issue
I was able to build the example in SoftConsole v6.5 and run it in the Renode emulator. However, I cannot get the "Run As" or "Debug As" choices to work. I don't see output on the serial terminals when I choose the "Run As" choice. When I try the "Debug As" choice I get an error which looks like it is related to a gdb command. 

### Running over JTAG
When I go to run the `mpfs-periodic-irq hw all-harts debug.launch` by right clicking on the file in the Project Explorer and choosing "Run As" and choosing the `mpfs-periodic-irq hw all-harts debug.launch` option I don't see any output on any of the terminals. 

The Eclicpse Console Output displays the following output: 
```
fpServer v17 waiting for incoming connections on the port 3334 with API v5
Device not found
Info : 1 1607540776825 microsemi_flashpro_server.c:1751 microsemi_flashpro_initialize() FlashPro ports available: S20153T6AV
Info : 2 1607540776825 microsemi_flashpro_server.c:1752 microsemi_flashpro_initialize() FlashPro port selected:   S20153T6AV
xPack OpenOCD (Microchip SoftConsole build), x86_64 Open On-Chip Debugger 0.10.0+dev-00859-g95a8cd9b5-dirty (2020-10-19-17:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
MPFS
Info : only one transport option; autoselect 'jtag'
Info : Hardware thread awareness created
do_board_reset_init
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : No Embedded FlashPro6 (revision B) devices found
Info : clock speed 6000 kHz
Info : JTAG tap: mpfs.cpu tap/device found: 0x0f81a1cf (mfg: 0x0e7 (GateField), part: 0xf81a, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 0 could not be made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: XLEN=64, misa=0x8000000000101105
Info :  hart 1: currently disabled
Info :  hart 2: currently disabled
Info :  hart 3: currently disabled
Info :  hart 4: currently disabled
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 1 could not be made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: currently disabled
Info :  hart 1: XLEN=64, misa=0x800000000014112d
Info :  hart 2: currently disabled
Info :  hart 3: currently disabled
Info :  hart 4: currently disabled
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 2 could not be made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: currently disabled
Info :  hart 1: currently disabled
Info :  hart 2: XLEN=64, misa=0x800000000014112d
Info :  hart 3: currently disabled
Info :  hart 4: currently disabled
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 3 could not be made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: currently disabled
Info :  hart 1: currently disabled
Info :  hart 2: currently disabled
Info :  hart 3: XLEN=64, misa=0x800000000014112d
Info :  hart 4: currently disabled
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 4 could not be made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: currently disabled
[... 231 lines cut ...]
When I go to run the `mpfs-periodic-irq hw all-harts debug.launch` by right clicking on the file in the Project Explorer and choosing "Debug As" and choosing the `mpfs-periodic-irq hw all-harts debug.launch` option I get the same output but when I set a breakpoint on something like line 46 of `e51.c` I get a "Problem Occurred" window. The contents of it say: 
```
Error in final launch sequence
Failed to execute MI command:
thread apply all set $pc=_start
Error message from debugger back end:
Unknown remote qXfer reply: 97370000 ,Failed to execute MI command:
tbreak e51
Error message from debugger back end:
Reply contains invalid hex digit 108
Failed to execute MI command:
thread apply all set $pc=_start
Error message from debugger back end:
Unknown remote qXfer reply: 97370000
Unknown remote qXfer reply: 97370000
Failed to execute MI command:
tbreak e51
Error message from debugger back end:
Reply contains invalid hex digit 108
Reply contains invalid hex digit 108
```

### Other Information

Host OS: Ubuntu 18.04

hughbreslin, 2020-12-10:
Hey Jesse 🙂 

Hmm this is an interesting one, I tried building and debugging this project and it connected straight away and I got a UART output. Its hard to see whats going wrong from the information you've got up so far:

1. You should be seeing something similar to whats shown here: https://youtu.be/BY3306ITDhM?list=PL9B4edd-p2ahc3AQdBAmGfAyo_pcDW0Tg&t=143 - does it just never get to the breakpoint at the e51 function for you to resume it?

2. If its never reaching e51, once you get to the last stages of the debug session starting can you halt the core and see where its currently executing? In the video I sent you can see the threads listed in the bottom left at 2:50 showing the current execution address - it'd be interesting to see where its getting stuck

3. Have you tried a different bare metal example and did that run? The mmuart-interrupt example is a good one to try

AntonKrug, 2020-12-10:
Could you try following?

- Go to Run-> Debug configurations
- Select that specific launcher and then go to the Debugger tab
- In the GDB Client Setup section where it has executable name "${cross_prefix}gdb${cross_suffix}" click on browse
- Navigate to the SC 6.5 RISCV gdb binary manually <SC_INSTALL_DIRECTORY>/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-gdb and select it instead of the macro
- Then click Debug button

Jesse-Millwood, 2020-12-10:
> Hey Jesse slightly_smiling_face
> 
> Hmm this is an interesting one, I tried building and debugging this project and it connected straight away and I got a UART output. Its hard to see whats going wrong from the information you've got up so far:
> 
>     1. You should be seeing something similar to whats shown here: https://youtu.be/BY3306ITDhM?list=PL9B4edd-p2ahc3AQdBAmGfAyo_pcDW0Tg&t=143 - does it just never get to the breakpoint at the e51 function for you to resume it?
> 
>     2. If its never reaching e51, once you get to the last stages of the debug session starting can you halt the core and see where its currently executing? In the video I sent you can see the threads listed in the bottom left at 2:50 showing the current execution address - it'd be interesting to see where its getting stuck
> 
>     3. Have you tried a different bare metal example and did that run? The mmuart-interrupt example is a good one to try

Thanks for your reply! For comparison's sake, I started the video from the beginning with the mmuart-interrupt example and followed up until the breakpoint happened on the e51. Mine does not seem to hit the breakpoint. When I click the suspend button in the top toolbar, I can see the `mpfs-mmuart-interrupt.elf [cores: 1,2,3,4,5]` line of the Debug tab in the lower left hand window has a pause icon on it. However, nothing pops up for me. I would expect a source file to pop up to show me where it is stopped the way it would if I did it with Renode. I do not seem to be able to actually halt it and see where it is and nothing prints on the my serial terminals. Do you have any other ideas? Thanks

Jesse-Millwood, 2020-12-10:
> Could you try following?
>
>     * Go to Run-> Debug configurations
>
>     * Select that specific launcher and then go to the Debugger tab
>
>     * In the GDB Client Setup section where it has executable [...]

@AntonKrug thanks for the tip! This sadly did not lead to any new output. This is what I see after I wait a couple minutes after hitting debug: 
[image]

Jesse-Millwood, 2020-12-10:
@hughbreslin and @AntonKrug , 
For what it's worth, I should also say that I can get Linux to boot from the SD card after I followed the instructions at [Updating the Icicle Kit](https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/boards/mpfs-icicle-kit-es/updating-icicle-kit/updating-icicle-kit-design-and-linux.md). So I don't think that my board is defective.

My main reason for going down this rabbit hole is that I can't seem to get the mss timer interrupt to fire when booting from the SD card. I am assuming it is the bootloader in the EMMC but I want to verify everything is working on the baremetal first. My code works on Renode when booting using the BBL so there is something different, I'm just trying to figure out what.

Jesse-Millwood, 2020-12-10:
@hughbreslin and @AntonKrug 
What is the state of your jumpers? Do they all follow the default settings in [the user guide](https://www.microsemi.com/document-portal/doc_download/1245042-ug0882-polarfire-soc-fpga-icicle-kit-user-guide)? What JTAG programmer are you using? I tried the onboard one and closing Jumper J9 but console reported an early failure that looked like it couldn't find it. Have you tried a FlashPro5 or FlashPro6?

Jesse-Millwood, 2020-12-10:
I actually just got it working! 
I had to run into the office to grab a FlashPro6 and that worked! 

It's odd that the Icicle Kit came with a FlashPro5. Using it as a JTAG debugger did not seem to work and neither did the onboard one.

closed 2020-12-10 by Jesse-Millwood

hughbreslin, 2020-12-10:
@Jesse-Millwood great glad its working🙂 

I haven't had any issues using a FlashPro5 but I'm on a Windows host so theres a difference there... You can get the onboard FlashPro6 working by following the steps in section 1.2 of the [Libero 12.5 SP1 RN](https://coredocs.s3.amazonaws.com/Libero/12_5_1/libero_soc_v12_5_1_release_notes.pdf)

Jesse-Millwood, 2020-12-10:
@hughbreslin Thanks for the pointer there. I'll have to see if I can get Libero working. My IT was having trouble with the FlexLM license working on our server before. I'll have to check and see we are still having those issues.

hughbreslin, 2020-12-10:
You actually only need flashpro express which doesn't need a license :) so you can avoid flexlm issues

Jesse-Millwood, 2020-12-10:
@hughbreslin thanks for your help! I just saw the Libero part and didn't read the rest. There seems to be an issue with using FPExpress and the job file but I have the external debugger working now so I'll just stick with that. I appreciate your help!

AntonKrug, 2020-12-11:
@Jesse-Millwood Depending on the eFP6 firmware on your board you might not even need flashpro express, just make sure J9 and J24 are properly set and it should work straight away (if you have a board with the newer eFP6 firmware)