polarfire-soc/polarfire-soc-bare-metal-library #3: Debug won't run: No Embedded FlashPro6 (revision B) devices found

jinchenglee, 2020-12-22:
I'm trying mpfs-mmuart-interrupt example following steps outlined in this [tutorial video](https://www.youtube.com/watch?v=BY3306ITDhM&list=PL9B4edd-p2ahc3AQdBAmGfAyo_pcDW0Tg&index=3&ab_channel=MicrochipTechnology). 

Because I have an early version of the board, it comes with the FlashPro5 JTAG Programmer/Debugger. I've already successfully activated eFP6 and updated it to rev B on an Icicle Kit, following steps shown in README.md in the https://github.com/polarfire-soc/icicle-kit-reference-design repository: 

_Ensure FlashPro Express is installed. It is installed automatically with Libero SoC or is available with the standalone Program and Debug tools
Download this FlashPro Express programming job
Follow the steps in this document to program the eFP6 on the Icicle Kit_

Also, I noticed from the tutorial video that _USB embedded programming connector_ is not enabled on my Icicle Kit board, so that I have to use the FlashPro5 JTAG debugger to connect to _JTAG programming header_ instead. I have also left open J21 and J9. But I still got these when trying to debug program:

_xPack OpenOCD (Microchip SoftConsole build), x86_64 Open On-Chip Debugger 0.10.0+dev-00859-g95a8cd9b5-dirty (2020-10-19-17:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
MPFS
Info : only one transport option; autoselect 'jtag'
Info : Hardware thread awareness created
do_board_reset_init
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : No Embedded FlashPro6 (revision B) devices found
'' does not match any available port_

jinchenglee, 2020-12-22:
BTW, my system is Ubuntu 18.04 LTS. Is it because I need FlashPro5 Linux driver?

My eFP6 update/activation was completed on Windows 10 PRO successfully.

jinchenglee, 2020-12-22:
I've run udev_install and /etc/udev/rules.d/70-microsemi.rules have been successfully installed. Which shows this:

```
SUBSYSTEM=="usb",ATTR{idProduct}=="2008",ATTR{idVendor}=="1514",MODE="0660",GROUP="4",SYMLINK+="FlashPro5"
SUBSYSTEM=="usb",ATTR{idProduct}=="6001",ATTR{idVendor}=="0403",MODE="0660",GROUP="4",SYMLINK+="FTDI232"
```
Still no luck. I still got the same error above. 

Interestingly, I can see one device from 'lsub':
```
jinli@jinli-Blade:/work$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0639:7213 Chrontel, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1532:0224 Razer USA, Ltd 
Bus 001 Device 004: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
**Bus 001 Device 006: ID 1514:2008 Actel** 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

hughbreslin, 2021-01-12:
Hey @jinchenglee! Sorry for the delay in getting back! Have you tried the steps in section 2.9.2 of [this doc](https://coredocs.s3.amazonaws.com/Libero/12_6_0/Tool/linux_setup_libero_soc_ug.pdf)? Theres an additional script to configure FlashPro 6.

jinchenglee, 2021-01-13:
@hughbreslin , as I mentioned at beginning, "I have an early version of the board, it comes with the FlashPro5 JTAG Programmer/Debugger." In this case, do I have the FlashPro6 on board usable? It seems section 2.9.2 of [this doc](https://coredocs.s3.amazonaws.com/Libero/12_6_0/Tool/linux_setup_libero_soc_ug.pdf) is about FlashPro6 only.

hughbreslin, 2021-01-13:
@jinchenglee hmm ok I'm not clear on what your issue is. I thought it was this:
1. You were using a external FlashPro5 as your embedded FlashPro6 was not activated
2. You activated the embedded FlashPro6 using the external FlashPro5 and programming file from the Reference Design repo
3. You tried to run a debug session from your Linux host using the embedded FlashPro 6 and it failed
4. You tried to run a debug session from a Windows host using the embedded FlashPro 6 and it works

Can you clarify where I've gone wrong? For my original answer I assumed you missed the set up steps for the udev rules required for the embedded FlashPro 6 which is what would have caused 3.

If you successfully ran the programming file linked in the reference design repository you have activated the embedded FlashPro 6 and made it usable.

hughbreslin, 2021-01-13:
And just to clarify after re-reading this "In this case, do I have the FlashPro6 on board usable? It seems section 2.9.2 of this doc is about FlashPro6 only." - there is no difference functionally between an embedded FlashPro 6 (on the Icicle Kit) or a standalone/external FlashPro 6 - they're all FlashPro 6 just different implementations. The implementation doesn't make any difference to the system set up required, i.e where this doc talks about "FlashPro 6" it is a coverall for whatever type of FlashPro 6 you could have.

jinchenglee, 2021-01-14:
@hughbreslin , sorry I wasn't clear enough. Let me reiterate:

1. I was using a external FlashPro5 as my embedded FlashPro6 was not activated
2. I (thought I have) activated the embedded FlashPro6 using the external FlashPro5 and programming file from the Reference Design repo. I thought it was successful because it either printed out "successful" or the result was just like what the tutorial had expected.
3. I tried to run a debug session from my Linux host using the embedded FlashPro 6 and it failed
4. I tried to run a debug session from my Linux host using the external FlashPro 5 and it failed too
5. I tried to run a debug session from a Windows host using the external FlashPro 5 and it works

My first question is:
1. Is there an embedded flahspro6 on my board? How do I make sure?
2. Why Linux failed even with external FlashPro 5?

hughbreslin, 2021-01-14:
@jinchenglee ah now I get you! 🙂 To answer your questions:

1. Yes - all of the Icicle Kits shipped with the parts for a FlashPro6. I say "parts" because FP6 isn't actually a dedicated ASIC, its a SmartFusion2 (SF2 is another one of our SoC FPGAs). This is how you could activate it using FlashPro express as you updated the programming on the SF2 part. You can see this on the kit, its between the programming header and the FP6 micro usb port:
[image]. Its not as clear in the [schematics](https://www.microsemi.com/document-portal/doc_download/1245126-polarfire-soc-icicle-kit-schematics) but you can see the header you have to open or close in the top right of page 10 along with the SF2 connections on pages 17 - 22. 

2. This I'm not sure of - have you ever been able to debug with Linux with any FlashPro? In 2 you mention doing the update with the FP5 but I'm wondering from what host as Linux seems to fail after that. Can you also let me know what distro you're using? I've tested this on Ubuntu 18.04 and 20.04 and I can use a FlashPro6 after following the steps in the doc I sent you on.

jinchenglee, 2021-01-15:
Thanks for replying. Let me try figure it out when I find some spare time.

jinchenglee, 2021-01-18:
@hughbreslin , I've tested again and here's the update. 

1. On Windows 10 Pro, I can debug with SoftConsole v6.5.0.442 using on-board FlashPro6. This did prove my on board FlashPro6 is working (at least under Windows 10). Below is the log:

2. On Linux (Ubuntu 18.04 LTS), I've followed section 2.9.2 of [this doc](https://coredocs.s3.amazonaws.com/Libero/12_6_0/Tool/linux_setup_libero_soc_ug.pdf), please see the contents of the two files installed into system:

```
$ cat /etc/cyusb.conf 

LogFile=cyusbd.log
PIDFile=cyusb.pid
<VPD>
1514 2009 Microsemi FP6 Programmer
</VPD>

$ cat /etc/udev/rules.d/88-cyusb.rules
KERNEL=="*", SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTR{idVendor}=="1514", MODE="666", TAG="cyusb_dev"
```

lsusb can find the device no problem, but the device is 1514:200b though, not 2009 as shown in /etc/cyusb.conf:
```
lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0639:7213 Chrontel, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1532:0224 Razer USA, Ltd 
Bus 001 Device 004: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 012: ID 1514:200b Actel 
Bus 001 Device 003: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Now when I try to launch debugging process in SoftConsole v6.5.0.422 (same version as installed on Window 10) under Ubuntu: it shows below:
```
xPack OpenOCD (Microchip SoftConsole build), x86_64 Open On-Chip Debugger 0.10.0+dev-00859-g95a8cd9b5-dirty (2020-10-19-17:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
MPFS
Info : only one transport option; autoselect 'jtag'
Info : Hardware thread awareness created
do_board_reset_init
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Embedded FlashPro6 (revision B) found (USB_ID=1514:200b path=/dev/hidraw7)
Error: Unable to open Embedded FlashPro6 (revision B) device
```

Good thing is that the FlashPro6 (revision B) can be detected. But it cannot be opened. I'm launching the SoftConsole SW using my normal account, shall I use sudo/root permission which I didn't try and isn't appreciated at all. 

My system info:
```
$ uname -a
Linux ComputerName 5.4.0-62-generic #70~18.04.1-Ubuntu SMP Tue Jan 12 17:18:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

jinchenglee, 2021-01-18:
"Good thing is that the FlashPro6 (revision B) can be detected. But it cannot be opened. I'm launching the SoftConsole SW using my normal account, shall I use sudo/root permission which I didn't try and isn't appreciated at all."

Ok, it turns out this is the key. I launched the SoftConsole using 'sudo' permission, now I can connect using FP6 on board and continue debugging process. 

I think I can close this issue now, but before I do that, is it possible to enable debugging on Linux w/o using sudo launching SoftConsole? That is really not a good habit on Linux. Thanks.

hughbreslin, 2021-02-15:
Hi @jinchenglee yes, check out page 21 of the SoftConsole [Release Notes](https://www.microsemi.com/document-portal/doc_download/1245416-softconsole-v6-5-release-notes) 🙂

nitindeshpande, 2021-05-06:
Hi,
 jinchenglee was able to flash the board with SC running with sudo permission and the extra steps to use SC in non-sudo mode are provided in SC Release note. I think we can close this issue.

nitindeshpande, 2021-10-07:
closing the issue since the resolution is provided.

closed 2021-10-07 by nitindeshpande