polarfire-soc/polarfire-soc-bare-metal-library #5: mpfs-sys-serv-example not running

rroulleau, 2021-02-15:
Icicle Kit Rev1.0
Using SoftConsole v6.6.0.463 I was able to run the mpfs-mmuart-interrupt following your readme with the 4 UART responding as expected.
https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/master/examples/Readme.md

I was hoping to run in the same setup the mpfs-sys-example to check you system services.
https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/tree/master/examples/mss-sys-services/mpfs-sys-serv-example
but I do not get the UART display_greeting() either in Polling or Interrupt mode. 
No error in SoftConsole, just lock in the while(1) loop waiting to get the command line on the UART0.
UART0 has the following settings MSS_UART_115200_BAUD, MSS_UART_DATA_8_BITS | MSS_UART_NO_PARITY | MSS_UART_ONE_STOP_BIT similar to the mmuart-interrupt.
Am I missing something obvious? Would you mind to give me a hint?
Thanks in advance.

hughbreslin, 2021-02-15:
Hey @rroulleau it looks like the clocks aren't being started in the system services e51.c file - can you try adding the following before uart init:

 SYSREG->SUBBLK_CLOCK_CR = 0xffffffff;
 
 I can see the clocks being started in the mmuart example but not in the system services one 🙁

rroulleau, 2021-02-15:
That's much better, thanks a lot! :-)

closed 2021-02-15 by rroulleau