polarfire-soc/polarfire-soc-bare-metal-library #7: `MSS_SYS_spi_copy` loses bits when configured for maximum speed.

archshift, 2021-03-15:
When using `flags=0` (80MHz), `MSS_SYS_spi_copy` will read from the SPI flash incorrectly. It will drop the last bit of every byte copied into RAM, and offset everything by one byte.

Here's a side by side example of read out vs ground truth bytes when this happens.

```
# MSS_SYS_spi_copy                                                                                          | # data.bin
00000400: 006f 6003 d800 0000 0000 1680 0000 0000  .o`.............                                         │00000000: dec0 07b0 0000 0000 002d 0000 0000 0000  .........-......
00000410: 006a 5d51 0b80 0000 0008 0300 0000 0000  .j]Q............                                         │00000010: d4ba a217 0000 0000 1006 0000 0000 0000  ................
00000420: 0068 1600 0000 0000 0000 0000 4000 0000  .h..........@...                                         │00000020: d02c 0000 0000 0000 0000 0080 0000 0000  .,..............
00000430: 0001 8000 0000 0000 007b 8000 0000 0000  .........{......                                         │00000030: 0300 0000 0000 0000 f700 0000 0000 0000  ................
00000440: 0000 0000 0000 0000 007b 0000 0000 0000  .........{......                                         │00000040: 0000 0000 0000 0000 f600 0000 0000 0000  ................
00000450: 0039 b697 b13a b4b6 3217 b836 30ba 3337  .9...:..2..60.37                                         │00000050: 736d 2f62 7569 6c64 2f70 6c61 7466 6f72  sm/build/platfor
00000460: b936 97b6 b833 3997 b334 b936 bbb0 b932  .6...39..4.6...2                                         │00000060: 6d2f 6d70 6673 2f66 6972 6d77 6172 652f  m/mpfs/firmware/
00000470: 97b3 3baf b830 bcb6 37b0 b217 32b6 3300  ..;..0..7...2.3.                                         │00000070: 6677 5f70 6179 6c6f 6164 2e65 6c66 0000  fw_payload.elf..
```

nitindeshpande, 2021-03-23:
Hello,
 The hardware actually limits the clock to 71MHz. 
 The driver and documentation needs an update. Will bring in the change in the next release.
 Thanks for pointing this.

nitindeshpande, 2021-08-12:
Hello,
 This has been resolved with the [polarfire-soc-bare-metal-examples v2021.05](https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/releases/tag/2021.05)