polarfire-soc/polarfire-soc-bare-metal-library #9: how to make printf() work

fcuzzocrea, 2021-05-12:
Hello!

I am trying to get some text printed out trough uart using print on a icicle board.

I am starting from the example here https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/master/examples/mss-mmuart/mpfs-mmuart-interrupt/src/application/hart0/e51.c, which is working (I can see the text sent over UART trough minicom) and I am just adding a     printf("Hello world!!\n"); right after line 111 (https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/master/examples/mss-mmuart/mpfs-mmuart-interrupt/src/application/hart0/e51.c#L111)

According to documentation in newlib-stubs.c (https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/8a970f7a995879530897b3aaf9cee2344ea3e3da/examples/mss-mmuart/mpfs-mmuart-interrupt/src/platform/mpfs_hal/newlib_stubs.c#L28) to route printf() output trough UART should be sufficient to add MICROCHIP_STDIO_THRU_MMUART0 #define. I have tried by adding it trough both trough SoftConsole and by directly adding it at the top of newlib_stubs.c. 
Doing so generates the following compilation error (https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/8a970f7a995879530897b3aaf9cee2344ea3e3da/examples/mss-mmuart/mpfs-mmuart-interrupt/src/platform/mpfs_hal/newlib_stubs.c#L57)

`../src/platform/mpfs_hal/newlib_stubs.c:59:50: error: 'g_mss_uart0' undeclared here (not in a function); did you mean 'g_mss_uart0_hi'?
 static mss_uart_instance_t * const gp_my_uart = &g_mss_uart0;
                                                  ^~~~~~~~~~~
                                                  g_mss_uart0_hi
make[1]: *** [src/platform/mpfs_hal/subdir.mk:61: src/platform/mpfs_hal/newlib_stubs.o] Errore 1
make: *** [makefile:51: all] Errore 2
"make all" terminated with exit code 2. Build might be incomplete.`

which I tried to solve by defining gp_my_uart as &g_mss_uart0_lo (like what is done on hart0/e51.c). Doing so made the file compile, but when I run it on the board and I attach minicom, the printf() output is not visible (what is printed trough the support function uart_tx_with_mutex() however is shown correctly).

What am I doing wrong?

Thanks!!

As a side note, here https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/8a970f7a995879530897b3aaf9cee2344ea3e3da/examples/mss-mmuart/mpfs-mmuart-interrupt/src/platform/mpfs_hal/newlib_stubs.c#L46 is stated that if no define is supplied the default baud rate is 57600 baud, however, here https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/8a970f7a995879530897b3aaf9cee2344ea3e3da/examples/mss-mmuart/mpfs-mmuart-interrupt/src/platform/mpfs_hal/newlib_stubs.c#L53 MICROCHIP_STDIO_BAUD_RATE is set to 115200 if no define is set.

hughbreslin, 2021-05-20:
Hey @fcuzzocrea we haven't done much work on printf to get it working and we've mainly been dealing with this using the uart prints. We'll come back and have a look at printf for a future release - is there a something specific you need printf for? I think the code you've found is a hold over from some older work

fcuzzocrea, 2021-05-20:
Thanks for your answer! Not really, was doing only some testing, but would be nice to have a working printf() implementation in the bsp, just in case someone wants to put some debug prints here and there in the code :)

hughbreslin, 2021-05-20:
thats fair 🙂 We might as well keep this issue open for people and I'll get back to you once we have an update

nitindeshpande, 2022-01-10:
The issue is being tracked against [platform](https://github.com/polarfire-soc/platform/issues/4) repo.

closed 2022-01-10 by nitindeshpande