polarfire-soc/polarfire-soc-bare-metal-library #11: main_first_hart() stucks on latest revision

fcuzzocrea, 2021-05-18:
Hi!

I am trying to simply print some strings trough UART using MSS_UART_polled_tx() and MSS_UART_irq_tx(), you can see my sample program here: https://gist.github.com/fcuzzocrea/76635d787df125a74ba3b5362f82f1ee , which basically is the same of the mpfs-mmuart-interrupt example located here: https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/master/examples/mss-mmuart/mpfs-mmuart-interrupt/src/application/hart0/e51.c .
I am using the bare metal library from master branch and compiling the following source files which are needed to use the aforementioned functions: 

`
    'pfsc-baremetal/src/platform/drivers/mss_mmuart/mss_uart.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_clint.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_irq_handler_stubs.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_l2_cache.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_mpu.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_mtrap.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_plic.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_pmp.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/mss_util.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_ddr.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_ddr_debug.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_ddr_test_pattern.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_io.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_nwc_init.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_pll.c',
    'pfsc-baremetal/src/platform/mpfs_hal/common/nwc/mss_sgmii.c',
    'pfsc-baremetal/src/platform/mpfs_hal//startup_gcc/mss_entry.S',
    'pfsc-baremetal/src/platform/mpfs_hal//startup_gcc/mss_mutex.S',
    'pfsc-baremetal/src/platform/mpfs_hal//startup_gcc/mss_utils.S',
    'pfsc-baremetal/src/platform/mpfs_hal//startup_gcc/system_startup_override.c',
    'pfsc-baremetal/src/platform/mpfs_hal//src/platform/mpfs_hal/startup_gcc/newlib_stubs.c',
    'pfsc-baremetal/src/platform/mpfs_hal//src/platform/mpfs_hal/startup_gcc/system_startup.c'
`
I am using the linker script located at

`
  pfsc-baremetal/src/platform/platform_config_reference/linker/mpfs-lim.ld
`

The source file is compiled using xpack gcc:

`
/opt/xpack-riscv-none-embed-gcc-10.1.0-1.1/bin/riscv-none-embed-gcc -v

Using built-in specs.
COLLECT_GCC=/opt/xpack-riscv-none-embed-gcc-10.1.0-1.1/bin/riscv-none-embed-gcc
COLLECT_LTO_WRAPPER=/opt/xpack-riscv-none-embed-gcc-10.1.0-1.1/bin/../libexec/gcc/riscv-none-embed/10.1.0/lto-wrapper
Target: riscv-none-embed
Configured with: /Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/sources/riscv-gcc-10.1.0-1.1/configure --prefix=/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/install/riscv-none-embed-gcc --infodir=/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/install/riscv-none-embed-gcc/share/doc/info --mandir=/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/install/riscv-none-embed-gcc/share/doc/man --htmldir=/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/install/riscv-none-embed-gcc/share/doc/html --pdfdir=/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/install/riscv-none-embed-gcc/share/doc/pdf --build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=riscv-none-embed --with-pkgversion='xPack GNU RISC-V Embedded GCC\x2C 64-bit' --with-bugurl=https://github.com/sifive/freedom-tools/issues --enable-languages=c,c++ --disable-mingw-wildcard --enable-plugins --enable-lto --enable-checking=yes --disable-tls --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-threads --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-riscv-none-embed --with-sysroot=/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-x64/install/riscv-none-embed-gcc/riscv-none-embed --with-native-system-header-dir=/include --with-abi=ilp32 --with-arch=rv32imac --disable-build-format-warnings --with-system-zlib
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.1.0 (xPack GNU RISC-V Embedded GCC, 64-bit) 
`
And uploaded to the polarfire soc throug GDB using a copy of openOCD extracted from the latest release of softconsole:

`
/opt/openocd/bin/openocd -v

xPack OpenOCD (Microchip SoftConsole build), x86_64 Open On-Chip Debugger 0.10.0+dev-00859-g95a8cd9b5-dirty (2020-10-21-21:16)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

`
The resulting binary however does not work as expected. In particular, by doing some trial and error with GDB I found out that it stucks in this while loop:

https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/8a970f7a995879530897b3aaf9cee2344ea3e3da/src/platform/mpfs_hal/startup_gcc/system_startup.c#L105

Commenting out line 105 to line 169 (so, skipping completely that while loop) kind of let the program run as expected.

Am I doing something wrong in my program?

Thanks!!!!

mallynch, 2021-05-18:
Hi Francescodario,

Sorry you are experiencing an issue. The platform folder contents of this example project are out of date. 
The currently released MPFS HAL in the platform in the src platform directory should resolve issue you are seeing.
There is a planned update of the example projects in the next two weeks.

In the interim can you replace the following directory contents
\src\platform\drivers\mss_mmuart
\src\platform\mpfs_hal
\src\platform\hal

With the ones from the src directory in github

https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/tree/master/src/platform
(commit)

Regards,
Malachy

fcuzzocrea, 2021-05-18:
Hi!! 

Thanks for your reply!

Actually, the code works correctly with the old  MPFS HAL (the one bundled in the platform directory of the example). The issue I am facing is when I try to use the latest MPFS HAL from the github master branch.

Thanks!!

nitindeshpande, 2021-08-12:
There has been a recent release on [polarfire-soc-bare-metal-examples v2021.05](https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/releases/tag/2021.05), could you please take look and see if your issues are addressed?

closed 2021-10-07 by nitindeshpande