polarfire-soc/polarfire-soc-bare-metal-library #13: mpfs_can_basic example hanging at MSS_CAN_config_buffer_n()

lucatonini, 2021-06-22:
Hi!

I am trying to test the CAN interface on an icicle board, using this example provided https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/master/examples/mss-can/mpfs-can-basic/src/application/hart0/e51.c

However when I try to run/debug the example through SoftConsole v2021.1-6.6.0.507 the program hangs when it tries to initialize Can 0 at 
https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/19494abb392447818ae585567f914d0611b232cf/examples/mss-can/mpfs-can-basic/src/application/hart0/e51.c#L104

When debugging with mpfs-can-basic hw all-harts I noticed that the program hangs and crashes at 
https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/19494abb392447818ae585567f914d0611b232cf/src/platform/drivers/mss_can/mss_can.c#L515 with the following debugger error.

```
Error: unable to resume hart 0
Error:   dmstatus =0x00030c82
Error:   was stepping, halting
Error: unable to halt hart 0
Error:   dmcontrol=0x80000001
Error:   dmstatus =0x00030c82
Error: Hart was not halted after single step!
Error: unable to step rtos hart
Error: unable to halt hart 0
Error:   dmcontrol=0x80000001
Error:   dmstatus =0x00030c82
```
Not sure why it is hanging, but could it be due to a miss config to the CAN devices registers or is there something else that I am missing to get the example to work?

Thanks in advance!

nitindeshpande, 2021-06-22:
Hello lucatonini,
May I know. which Libero design is programmed on your icicle kit?

nitindeshpande, 2021-06-22:
And, are you using the .xml which is provided with the mpfs_can_basic project or have you modified/replaced it?

The base address of the CAN0 is influenced by the setting in the APBBUS_CR bit 14. Could you check if that register is getting modified somehow?

lucatonini, 2021-06-22:
Hi,
I am using the .xml that was provided in the mpfs_can_basic example.

nitindeshpande, 2021-06-22:
Ok. Please check the APBBUS_CR register value and that might give some hints.

lucatonini, 2021-06-22:
During the whole life cycle of the program, the APBBUS_CR bit 14 is set to 0. Which is correct according to the PF_SoC_RegMap_V1_2 documentation  that says it should be set to 0 for the Default Base Address. As can be seen in the screenshot below.

[image]

And only if the `g_mss_can_0 = &g_mss_can_0_hi` as seen in the screenshot below, then I am able to successfully run the program.

[image]

If `g_mss_can_0 = &g_mss_can_0_lo` as seen in the screenshot below, then the program hangs and crashes.

[image]
 
Right up till the point that the program crashes the APBBUS_CR bit 14 is set to 0.

Any Idea on why the APBBUS_CR bit address does not match that of the CAN0 Default Base Address?

nitindeshpande, 2021-06-22:
Thanks for the details.
I was able to reproduce the issue you see.
I programmed the board with latest released reference design and used the mpfs-can-basic project without changing the .xml in it, and i could see the crash happening at line 515 of mss_can.c
It looks like the issue is due to mismatch between the design that is running on the board and the design description xml being used in the SoftConsole project.
If you use the latest .xml into your SoftConsole project you will see it working with g_mss_can_0 = &g_mss_can_0_lo.

Please note that when you update the to latest design/.xml, you should also update the MPFS HAL in your SoftConsole project with the latest available version from the [platform](https://github.com/polarfire-soc/platform)

Alternatively, we are presently working on updating the example projects, you might want to use them directly once they are available. Planned for this week.

nitindeshpande, 2021-08-12:
Easy to test CAN loopback example is released with [polarfire-soc-bare-metal-examples v2021.05](https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/releases/tag/2021.05).

nitindeshpande, 2021-09-03:
@lucatonini, Were you able to test the latest released projects? Is this issue be closed