polarfire-soc/polarfire-soc-bare-metal-library #16: I2C Status Register Values

kowalski100, 2021-08-12:
I did a quick match of I2C_STATUS register (attached) with values given in driver [here](https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/6a093aa9570180170a3874ecbed78a658666cf7d/src/platform/drivers/mss/mss_i2c/mss_i2c.c#L38). The register doesn't mention all the values mentioned in the driver. Where are other values like **ST_SLAR_ACK** taken from?

[image]

kowalski100, 2021-08-25:
Any updates on this?

nitindeshpande, 2021-09-03:
@kowalski100 
Confirming the issue. The status register has different values based on the current mode of operation. The register description only describes the values in Master Transmit mode. Values when the core is in other modes are documented.
Will get this corrected. Thanks for pointing out.

kowalski100, 2021-09-07:
Hi @nitindeshpande Thanks for your reply.

Closing this issue and waiting for your reply confirming documentation update.

closed 2021-09-07 by kowalski100