polarfire-soc/polarfire-soc-bare-metal-library #17: I2C bit rate settings

kowalski100, 2021-09-07:
Hi, 

I am looking on setting bit rate of I2C driver. I don't see any bit rate settings except I2C0_CTRL register as shown in figure below (highlighted).

[image]

Suppose I want bit rate of 100kb/sec (the standard i2c bit rate) and input clock frequency is 150Mhz, what will be my settings?

nitindeshpande, 2021-09-08:
Hi,
 1) You will unfortunately not be able to achieve exact 100KB/sec with 150Mhz. The closest you will be able to get is by choosing '100', that is devisor value of 960. 
     I am not sure about the value '111' though, because I am not sure what is the source of BCLK. I will get back to you as soon as I get to know.

2) There is no explicit 'master mode' / 'slave mode' setting via registers.
    The Core infers the mode via from its present state and the combination of ENS1, STA and AA bit values. The MSS I2C driver handles it for you.

Hope that helps.

Keval-Joshi, 2021-09-09:
Hello, 

The required registers information has been provided via email and the register map description will be updated in due time. 

Thanks.

closed 2021-10-07 by nitindeshpande