polarfire-soc/polarfire-soc-bare-metal-library #18: I2C SCL line remains high

kowalski100, 2021-09-08:
Hi,

I wrote a baremetal application to test I2C. I am doing it in polling mode by manually sending start condition then address. On J44 i connected SCL and SDA lines to oscilloscope. Following is code snippet:

`
/*some initialization code like baudrate etc...*/
....
I2C_SEND_START();
while (I2C_READ_STATUS() != ST_START); // wait till start condition is sent
I2C_SEND_ADDR(start_byte); // send address with r/w bit.
`

**What I see on oscilloscope:**
1. as soon as I hit `I2C_SEND_START()` **SDA** get low and remains low and **SCL** remains high (a slight glitch but doesn't get low). status register indicates that start condition is sent.
2. No clock is observed on oscilloscope.
3. No address is sent although I place it in DATA register after start condition as shown in above code snippet.

Following is waveform is see on Oscilloscope. **Note that no slave is connected.**

                                      glitch
SCL ----------------------,/------------ (high)

SDA ------------------|          
                                  |
                                  |..............................................................

kowalski100, 2021-09-08:
I have almost similar case as shown below. 
Yellow: SCL
Magenta: SDA

[image]

Note: I took the image from the following post.
https://silabs-prod.adobecqms.net/community/mcu/32-bit/forum.topic.html/i2c_bus_stuck_withsdalowandsclhigh-GIEz

Keval-Joshi, 2021-09-09:
Hello, 

This issue looks similar to #17. The required information has been provided over there.  

Thanks.

kowalski100, 2021-09-09:
The issue is still there. SCL line stays high. Though the status register value changes that Address has been sent but not visible on oscilloscope.!!!

Keval-Joshi, 2021-09-09:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hello Ijaz,

Could you please try MSS_I2C bare-metal example project and observe the clock signal on oscilloscope. ?
You can find mss_i2c example projects here<https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/tree/main/driver-examples/mss/mss-i2c>.
Please use mss-i2c-pac1934-sensor for the same as mss-i2c-master-slave project require separate loop back design.

Thanks & regards
K.Joshi

kowalski100, 2021-09-09:
I ran mss-i2c-pac1934-sensor example and changed I2C1 to I2c0 just to check sda and scl on oscilloscope. Nothing is visible. with I2C1 it is working fine.

hughbreslin, 2021-09-14:
Hey @kowalski100 did you base your application off a released example? The last release for this repository was 2021.02 but support for I2C0 was only added to the reference design / XML in 2021.04 - if you just pulled the zip of the 2021.02 release you'd get older [XML](https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/2021.02/examples/mss-i2c/mpfs-i2c-pac1934-sensor/src/boards/icicle-kit-es/soc_fpga_design/xml/ICICLE_MSS_0.xml#L950) which doesn't have I2C0 configured whereas pulling from main will get newer [XML](https://github.com/polarfire-soc/polarfire-soc-bare-metal-library/blob/master/examples/mss-i2c/mpfs-i2c-pac1934-sensor/src/boards/icicle-kit-es/fpga_design/design_description/ICICLE_MSS_eMMC_cfg.xml#L1238) - this could explain your issue.