polarfire-soc/polarfire-soc-bare-metal-library #20: Getting errors from the example project "\\polarfire-soc-bare-metal-library\examples\mss-rtc\mpfs-rtc-time"

henryding-emcore, 2022-03-11:
Hello, 

I'm using this example project to do some FIC1 access where MSS is the master and the Fabric is the slave. 
The lines I added to u54_1.c are after "MSS_RTC_clear_irq()":
uint64_t *pointer2;
pointer2 = 0x3000000000;
*pointer2 = 0x8877006605500000;

The following errors are what I got after I terminate the debug:
Error: unable to halt hart 1
Error:   dmcontrol=0x80010001
Error:   dmstatus =0x00030c82

I'm able to see this writing happens on the fabric. But for some reason, the processor crashes with these errors.

hughbreslin, 2022-04-28:
Hi @henryding-emcore sorry for the delay, unfortunately this repository has been deprecated, we are now using the [bare-metal-examples](https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples) repo. 

In regards to this issue - have you taken the fabric out of reset? When you say you can see the writes on the fabric are they succeeding? Based on the address you've added some logic to the reference design on FIC1, if you've based your resets on the Icicle Kit Reference Design you will also have to set the fabric reset bit, it could be the case that the FIC itself is not in reset but the fabric peripherals are. The error you've copied is what occurs when a hart tries to access an illegal address or a peripheral in reset where the access doesn't return and the hart hangs waiting for it. 

Please let me know if you're still facing this issue. You can enable the fabric using the following:
`SYSREG->SOFT_RESET_CR   &= (uint32_t)~(SOFT_RESET_CR_FPGA_MASK);`

henryding-emcore, 2022-04-28:
> [EXTERNAL EMAIL] DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

Thanks, Hugh!

All the resets are released. It was fixed by adding a COREAXI4INTERCONNECTOR.

Thanks,
Henry

This email, its contents and attachments contain information from EMCORE Corporation and/or its affiliates which may be privileged, confidential or otherwise protected from disclosure. If you are not the intended recipient, you may not read, copy, distribute, or use the contents of this message. If you have received this email in error please notify the sender by reply e-mail and delete the original message and any copies.

closed 2022-05-03 by Cyril-Jean